LibreScanner/ciclop #5: Compatibility with Arduino Uno

jbernardis, 2015-02-08:
Is this compatible with the arduinio Uno, or is the BT-328 mandatory.  It is hard to tell, but it does not seem as if you are using any of the features of the BT-328 - which I hope makes it compatible with Arduino.  I have no objection to the BT-328, but they're a bit more expensive than the arduino, and hard to come by here in the USA.  In fact, I see only 1 vendor in the UK

Jesus89, 2015-02-09:
Hi!

It is designed to be used with BT-328, because of the 4 square holes and BT connectivity. Arduino Uno fits but you need to redesign camera holder.

jbernardis, 2015-02-10:
Your answer is a little ambiguous.  The statement about redesigning the camera holder leaves open the possibility that this will work with an arduino.  So let me be explicit in my question.  Will the firmware for this scanner function on an Arduino?  By that I mean will it operate the lasers and stepper motor in concert with the PC code so that scanning can take place?  Or do I need to purchase a BT-328.  I know the BT-328 has bluetooth circuitry, and it looks pretty cool, but I don't see where you're using bluetooth.  Beacuse of that, I thought perhaps the Arduino will work instead.

MMJeon, 2015-02-10:
Hi. Jesús 

I like to use the Arduino Uno & GRBL Shield. But, I do not know the connection method. 

Pinmap the Arduino Uno is required.

Your help is needed.

jorgerobles, 2015-02-10:
@jbernardis, Ciclop is fully compatible with Arduino Uno and other Atmel328. I've got Horus-FW running on a Nano V3 Board. Bluetooth is not needed for now, but you could pair with your computer to substitute a wire.

jorgerobles, 2015-02-10:
@MMJeon.  VicentM at [https://groups.google.com/forum/#!forum/asrob-uc3m-impresoras-3d](Clone Wars Google Group) has improved the pin map to use the CNC Shield V3 for Arduino Uno. [https://groups.google.com/group/asrob-uc3m-impresoras-3d/attach/1eb669124868fe4f/cpu_map2.h?part=0.1&authuser=0&view=1](cpu_map.h)

MMJeon, 2015-02-10:
jorgerobles. 

Thank you. The problem was solved thanks.

Jesus89, 2015-02-10:
Thanks @jorgerobles for your correct replies! :D

closed 2015-02-10 by Jesus89

jorgerobles, 2015-02-10:
My pleasure!

CashYue, 2015-03-07:
The connection is fail.
Since I have the same problem, I would want to know the solution.

MMJeon, 2015-03-09:
Please check bauds. My installed Horus_FW runs at 115200 baud

avsbot, 2015-03-19:
I use Arduino UNO, everything works fine. A Bluetooth does not save, the camera still has a cable))