LibreScanner/ciclop #6: Horus and Arduino Uno error.

MMJeon, 2015-02-11:
Hi. 
Ask for help with the following problem.

* Upload Firmware is an Arduino Uno was in Horus.  The problem looks like the picture below.

* OS : Windows 7 home Premium 32bit
  Hardware : only Arduino Uno, LogiTech C270
  
* Image links attached.

 http://postfiles10.naver.net/20150211_25/taifight_14236486486233DDS2_JPEG/0001.jpg?type=w1

 http://postfiles1.naver.net/20150211_224/taifight_1423648648840DwjMW_JPEG/0002.jpg?type=w1

 http://postfiles7.naver.net/20150211_134/taifight_14236486489870A8Mq_JPEG/0003.jpg?type=w1

 http://postfiles5.naver.net/20150211_36/taifight_1423648714846cnAXB_JPEG/20150211_184859.jpg?type=w1

jorgerobles, 2015-02-11:
Please check bauds. My installed Horus_FW runs at 115200 baud

MMJeon, 2015-02-11:
jorgerobles. Thank you.

Was a success.

closed 2015-02-11 by MMJeon