LibreScanner/ciclop #9: About the board

CashYue, 2015-03-07:
Hi
I'm wondering whether I could use an arduino Uno and a CNC shield V3 to make one.
If it is possible, what should I change in software and hardware?

Jesus89, 2015-03-11:
Hi! It is possible but it is not implemented yet. You probably need to change pin map in firmware (https://github.com/bq/horus-fw). If we do that in the future, we will notify you.
Regards!

Minkynl, 2015-03-12:
Is the firmware already suited for the combination of Arduino Uno + ZUM SCAN?

Jesus89, 2015-03-12:
Yes! You can also burn the firmware from the application https://github.com/bq/horus.

CashYue, 2015-03-12:
You said that the zum scan is based on cnc shield on your website.
What's different?
Where can l find the pin map or I need to try by myself now?

Jesus89, 2015-03-12:
ZUM Scan has laser control (with transistors), also direct analog sensor connectivity and Jack connector.
You can find all the sources here: https://github.com/bq/zum/tree/master/zum-scan.

closed 2015-03-25 by Jesus89