LibreScanner/ciclop #20: Pattern Surface and Calibration Checkerboard quest

FLIPDAVE, 2015-11-24:
I'm confused over the dimensions of the pattern surface, and the calibration checkerboard? Could you please provide those dimensions for the plexiglass patttern? Also,the size of the checkboard pattern; does the block size of the checkerboard make a difference related to the calibration process?

Dave

nallath, 2015-11-25:
It uses the defaults of OpenCV. In case of the size for a checkerboard, the OpenCV implementation uses the distance between the corners as the ¨object size¨.  

There is no clear answer to if more blocks or a larger size gives better calibration results. In a perfect situation you´d have a calibration board that fills the entire view of the camera. This is usually not possible, so to make it more robust, the pattern tends to be smaller.

Larger blocks (eg; less blocks in the same area) tends to improve the sub pixel accuracy of the corners, but more blocks gives you more points to calculate the calibration.

jorgerobles, 2015-11-25:
It's on https://github.com/bqlabs/ciclop/tree/master/svg,
Open with inkscape, as it has different pixels per inch than illustrator. The plexiglass is 170x120x3 mm

Bantum-Works, 2015-11-26:
Found the checkerboard measures in imperial : 1/2 inch squares ... ;)
You could modify it to metric & go 1 cm square - But would it stuff up the scale of objects in Horus ? 
As my understanding of the software it uses imperial measurements & calibrations are based on that ... 
Can anybody confirm this ?

nallath, 2015-11-26:
For openCV it doesn't matter. If you give the size in imperial, the calibration is in imperial and therefore the scan results are in imperial. Same goes for metric.

closed 2016-01-11 by Jesus89