LibreScanner/ciclop #21: LDR & Arduino Sketches

Bantum-Works, 2015-12-28:
Hi - I notice in the Arduino sketch & on Horus for the Ciclop Scanner it mentions LDR ...

My query relates to what the bqlabs specifications call for ( type of diode ), what's required for its use  ( suitable location ) & any means ( notes or otherwise ) of calibrating them into system.

As my knowledge of light sensitive diodes is limited, the basic understanding is that they register the amount of light received & the program can then adjust the amount needed by a camera.

Also are there any plans to update the program to include the LED lights into the Arduino Sketch ?
As I'm thinking of using a track ball to control them via program found on fritzing : http://fritzing.org/projects/control-rgb-led-strip-using-ps2-trackball-arduino 

I'm in the process of finishing a scanner & adding some other tweaks ... 
See here for more details : http://www.thingiverse.com/thing:954056

Much appreciate some assistance in the matter.

Regards,
Bantum Works ...

Jesus89, 2016-01-11:
Hi, great job in your evolution :D

Light and LDR for Ciclop needs research in future, also new hw and a light calibration system.

For scanning, the only requirement for background is to be static, no matter what colour. Best objects to scan should be opaque and matte. Also dark objects are more difficult to scan than light coloured objects.

Regards.

Bantum-Works, 2016-01-23:
Thanks,

Will look at using some black felt & see how that goes ... :)

As to the RGB LED's - intent is to be able to adjust their colour to get best match for camera to pickup subtleties in surface, as from what I understand some cameras pick up certain colours better than others. Will have a bit of a play to see how this goes ... ;)

My other query relates to how to add / integrate the Arduino sketches ... got any tips on how ?

Cheers, Bantum ...

Jesus89, 2016-01-25:
Hi,

You can find the current firmware here: https://github.com/bqlabs/horus-fw.

Regards.

Bantum-Works, 2016-03-12:
Bit of an update :

I've tried the black felt background, but found was too dark for camera - so ended up going for a white felt instead, you can see photos on the Thingiverse : http://www.thingiverse.com/thing:954056

Haven't been able to find a workaround with the LED's as yet, so will need to do some research & find out more about LDR, got any suggestions ? 

More to come ... :/

Ciao, Bantum ...

Bantum-Works, 2016-06-04:
Hi, Thanks for the update - am looking into making it all work eventually.

One other query has come up - I'm also trying to use Logitech's c920 with HD 1080 camera, Does the Horus software enable this higher resolution ? or am I wasting my time, as I can't get any decent scans out of it atm. so not sure weather to keep it or go back to original camera ... :/

Regards,

Karl Schulze ...