LibreScanner/ciclop #38: calibration pattern rarely registers

BETLOG, 2017-04-14:
The calibration checkerboard is insanely unreliable.

It seems like unless lighting conditions are absolutely perfect - with a soft and omnidirectional diffuse light - that the calibration pattern is invisible to Horus.
I have tried using a light tent and a multitude of different types of LED's in every conceivable position and intensity, and yet the calibration pattern frequently drops out, even viewed front-on, or simply never registers.
Increasing capture contrast seemed to help for a while, but this also drastically reduces it's trackability on more incidental angles. However now I can't get it to register at all no matter how I adjust settings or accept defaults.

Everything else about the machine is working perfectly. I have put it together and fine tuned it three or four times now, and have previously managed to produce some excellent scans, so it's frustrating that I can't even get it to see the calibration board any more.

Why is it so unpredictable?
Are my black squares not black enough?
Do the imperfections in my photocopy effect it's usefulness?
Could there be some issue with 50/60Hz flickering?.. or spectral inadequacies of warm and cool LEDs? 
Is deleting ~/.horus not enough to *ensure* that *everything* is reset to default? 

Also: What is the best form of calibration board? I have considered 3d printing one... or having one professionally printed on thich card stock. What is the perfect form?

And no, removing the rubberband at the bottom of the board makes no difference whatsoever.
[image]

leccybill, 2017-04-16:
Hi check the camera intrinsic sin advanced calibration. I've given this scanner bad reviews everywhere but if you manually rotate the turntable and press whatever your version says to image it then manually move turntable if it accepts it will turn green for you to accept. Do that through 180 degrees and your scans should hopefully improve. Yes I agree nowhere does it sat how important this step is. I've got great scans now but been trying to use Meshlabs and it is a complete mystery how my meshes get recognised. So will look forward to any pointers. I may even write a new manual on how to get this scanner working but if you check the history. I'm  experimenting with Rasberry Pi and  Fabscan Pi. Hope that helped.

BETLOG, 2017-04-17:
The scanner design is quite sound. 

It's something in the software; a threshold where pattern detection fails.

Maybe we should be prodding the OpenCV community to rectify this.. I'm not sure.. but i'd have normally expected this community to at least echo our complaint in this case. It's difficult to imagine why everyone isn't having this issue.

What imaging library does fabscan use?

But yes; the documentation seems to fail to highlight the importance of a few things. Probably the most significant for people having a poor experience is lighting. As a photographer I am already aware that ideal lighting is necessary for quality reproductions of anything, however in this case the acceptable margin for error is ridiculously small.

I'm not interested in manually forcing it to recognise the calibration plate and getting a calibration with holes in it. I want it to recognise the plate as easily as it should and getting a perfect calibration, so I can make perfect scans.