LibreScanner/ciclop #39: Camera not Recognised

Dbutler431, 2017-04-14:
Scanner has been working for a few days and gradually getting better scan results, however, I can not get the scanner to connect and get a message to say that the camera has not been detected. Webcam shows up in list of video devices available, but keep get a message that the wrong camera has been selected. Does anyone know how to resolve this issue.

leccybill, 2017-04-16:
Hi are you using an IMac? I had loads of trouble with conflicts with internal camera and external add on typical Apple incompatibility issue if that's what you're suffering get an old laptop and load Linux or sorry but use Windows those were the only solutions I found. Hope that helps

Dbutler431, 2017-04-17:
I am using Windows. Everything was working ok, but I think there was an upgrade to Windows and the camera stopped working on restart.